sharebrained/portapack-hackrf #10: Retain settings when power is disconnected

jboone, 2015-07-08:
Write data to RTC battery backed RAM (simple but limited storage), or to SD card (not so simple, risky).

jboone, 2015-07-09:
Of course, there may be no SD card, it may be non-functional, or ejected at any time. So solving the problem with the RTC RAM is preferred, even though it's tiny (8Kbytes, I believe).

aernan, 2015-08-02:
oh. whoops.

jboone, 2015-08-24:
Lesson learned: there's only 256 *bytes* of RAM ("REGFILE") at 0x40041000, as part of the RTC peripheral, that is maintained when power is removed from the PortaPack. Still, that's plenty to store important state, if done carefully.

jboone, 2015-08-24:
Code to use RTC REGFILE for tuning frequency and PPM value seems to be working well. Closed, though the implementation could use improvement, e.g. versioning the persistent data and clearing or migrating the data when the version ID is incremented.

closed 2015-08-24 by jboone