sharebrained/portapack-hackrf #12: Ubuntu 15.04 ARM GCC firmware compile failure

sharebrained, 2015-07-08:
Compiling main.cpp
In file included from main.cpp:41:0:
dsp_decimate.hpp: In constructor 'dsp::decimate::FIR64AndDecimateBy2Real::FIR64AndDecimateBy2Real(const std::array<short int, 64u>&)':
dsp_decimate.hpp:63:18: error: invalid initialization of reference of type 'const std::array<short int, 64u>&' from expression of type '<brace-enclosed initializer list>'
  ) : taps { taps }
                  ^
main.cpp: In constructor 'FSKProcessor::FSKProcessor(MessageHandlerMap&)':
main.cpp:577:42: error: invalid initialization of non-const reference of type 'MessageHandlerMap&' from an rvalue of type '<brace-enclosed initializer list>'
  ) : message_handlers { message_handlers }
                                          ^
../chibios/os/ports/GCC/ARMCMx/rules.mk:165: recipe for target 'build/obj/main.o' failed
make[1]: *** [build/obj/main.o] Error 1
make[1]: Leaving directory '/home/__redacted__/portapack-hackrf/firmware/baseband'
Makefile:89: recipe for target '../baseband/build/baseband.elf' failed
make: *** [../baseband/build/baseband.elf] Error 2

sharebrained, 2015-07-08:
The relevant package is: arm-none-eabi-gcc (4.8.3-18ubuntu2+12) 4.8.3

jboone, 2015-07-17:
This error is correct by the letter of the C++11 standard. Read:

http://stackoverflow.com/questions/10509603/why-cant-i-initialize-a-reference-in-an-initializer-list-with-uniform-initializ

However, it was considered an oversight in C++11 and has been changed in C++14:

http://open-std.org/JTC1/SC22/WG21/docs/cwg_defects.html#1288

GCC 4.8 is correct to emit this error, by C++11 standards.
GCC 4.9 is correct to NOT emit this error, by C++14 standards.
So what to do?!

closed 2015-07-17 by jboone