sharebrained/portapack-hackrf #2: Improve firmware update experience

jboone, 2015-07-08:
Especially for Windows, but other platforms could use more love, too.

KC5CQW, 2015-08-26:
Not much luck here compiling from src on OS X...

jboone, 2015-08-26:
Can you provide details? I am developing on OS X 10.10.5, so it *should* work just fine.

KC5CQW, 2015-08-26:
I couldn't get the ARM compiler to instal on my X.9 system.

jboone, 2015-08-26:
OK, moving this to new issue #60. Please follow up with more details there.

jboone, 2015-09-03:
HackRF Mode in the new firmware will help a bit with this, eliminating the need for DFU mode except in dire circumstances where the firmware on the device is non-functional.

jboone, 2015-10-22:
[Windows firmware tool](https://github.com/sharebrained/portapack-hackrf/wiki/Updating%20Firmware%20Windows) is now available.

closed 2015-10-22 by jboone