sharebrained/portapack-hackrf #5: Improve power consumption

jboone, 2015-07-08:
Review LPC4320 peripherals and clocks that can be shut down because they aren't in use.
Check that peripherals are turned off if they're not used for a while (e.g. SD card).
Review ChibiOS code, ensure LPC4320 cores are sleeping when they can.
Align interrupts, threads to reduce the number of interrupts. Particularly on the baseband processor, where RSSI/touch sampling blocks can be aligned with baseband sampling. This will also help align timestamps between RSSI and baseband sample.

JDogHerman, 2015-08-06:
Would it be possible to turn off the backlight of for the screen to save power?

KC5CQW, 2015-08-06:
And maybe a dim adjustment feature for when full brightness is not needed.

jboone, 2016-01-28:
Added display sleep mode, which drops power by 60-70mA measured the 5V USB input. Lots more possibilities, including stopping spectrum calculation and various baseband/RSSI/audio/process stats, which is wasted effort when display is not on.