sharebrained/portapack-hackrf #7: Digital modes

jboone, 2015-07-08:
Break out digital demodulation/decoding into a separate mode?

aernan, 2015-08-02:
I could see workflow of scanning bands to find a signal. Zooming in to see the signal then attempting different demodulation techniques. If you can't easily demodulate then capture for offline demodulation.

jboone, 2015-08-02:
aernan: That's exactly what I was hoping to accomplish. Right now, I'm imagining a zoom button that would toggle between channel bandwidth and maximum (20MHz) bandwidth. Demodulation would be suspended and audio muted during the 20MHz mode, while you identify a signal of interest and retune for it. Then, you'd zoom back in to the channel and demodulation/audio would resume.

aernan, 2015-08-02:
> aernan: That's exactly what I was hoping to accomplish. Right now, I'm imagining a zoom button that would toggle between channel bandwidth and maximum (20MHz) bandwidth. Demodulation would be [...]
That sounds perfect. I would totally eat that up.

KC5CQW, 2015-08-05:
Clarification; is digital demodulation to take place "off-board"? This would indeed be quite taxing and processor intensive for local display...
Yes, I would also utilize the zoom/tune feature.
Hmm, that gives me an idea... What about an Auto Frequency Correction with a marker icon or +/- Hz/kHz delta for tracking a drifting signal? This would be focused on the carrier or peak amplitude components of the signal.

aernan, 2015-08-06:
To clarify. I'm guessing some forms of digital signals are too complex to demodulate on the portapack. Things like GSM and Wifi. So if you can't (due to horsepower) then make it easy to capture them so you can then use another host (desktop/laptop) to do the heavy lifting. For everything else do best effort to demodulate on the portapack.

JDogHerman, 2015-08-06:
It would be nice to have a decoder of some basic digital modes on-board. DTMF, RTTY, CW, PSK31

JDogHerman, 2015-08-06:
Or maybe decode WFM HD RDM signals.

KC5CQW, 2015-08-06:
Being Linux based, I wonder if the HackRF/PP H1 could utilize some of the fldigi source. 
http://www.w1hkj.com/usercontrib/FLDG-2013-10-01-2111.zip

jboone, 2016-07-05:
@KC5CQW You're no doubt aware of this by now, but the HackRF and PortaPack are not capable of running Linux -- there's far too little RAM (200K) to consider such a thing.

jboone, 2016-07-05:
This issue wound up becoming very broad. Moved digital mode ideas to issue #89. Zoom out idea moved to issue #90. Spectrum peak marker moved to issue #91.

closed 2016-07-05 by jboone