sharebrained/portapack-hackrf #16: Provide waterfall in receiver WFM mode

jboone, 2015-07-09:
Code needs to be restructured to do fewer FFTs due to less available CPU cycles. So time-domain coverage will be poorer than in other modes. But it'd still be useful to have...

aernan, 2015-08-02:
I would use this feature. Even if you could not see as much of the spectrum.

KC5CQW, 2015-08-05:
Useful for visualizing modulation drive and deviation. It need not have a high resolution as most WFM signals are audio only. The display refresh can be in the order of seconds or long enough for an average spoken sentience to be graphically populated on the screen. (5-10 sec.)

aernan, 2015-08-06:
Many WFM signals have a HD Radio signal on them as well. Would be nice to see the waveform to see if HD is there. http://hdradio.com/ https://www.ibiquity.com/technology

KC5CQW, 2015-08-06:
Agreed! This is a basic feature found in GQRX.

jboone, 2015-08-07:
Quick implementation in "wfm_waterfall" branch, but scrolls too quickly and hangs sometimes. A debugger will be necessary, and I don't have one on-hand at the moment.

jboone, 2015-08-07:
Initial filtering and decimation steps may need to be tweaked. CIC filter characteristics seem to be visible in the waterfall, which reduces its utility.

jboone, 2016-01-05:
Duplicate of #63, which is now closed.

closed 2016-01-05 by jboone