sharebrained/portapack-hackrf #24: Adapt bootstrap/Makefile to work with dfu-util 0.8.

jboone, 2015-07-18:


jboone, 2015-07-30:
"make program" target is now in firmware/Makefile, arguably where it belongs.

aernan, 2015-08-02:
the workaround is:
dfu-util --device 1fc9:000c --download hackrf_one_usb_ram.dfu --reset
hackrf_spiflash -w portapack-h1-firmware.bin

ZeroChaos-, 2015-08-03:
https://github.com/sharebrained/portapack-hackrf/blob/master/firmware/Makefile#L55

isn't that what it already does? I don't understand how this is a "workaround"

jboone, 2016-07-28:
At some point, I made this work, or it fixed itself. I've been using dfu-util 0.8 for quite a while now. It complains about not having a valid DFU suffix, or the CRC not matching, which seems like a different issue. Closing this.

closed 2016-07-28 by jboone