sharebrained/portapack-hackrf #27: Establish version/release schedule, naming convention, location for binaries, etc.

jboone, 2015-07-26:
You know, proper build and release engineering...

ZeroChaos-, 2015-07-27:
suggestion:  two scripts that handle the firmware loading for the user.  "switch_to_portapack" and "switch_to_hackrf" would be naming suggestions.

ZeroChaos-, 2015-07-27:
suggestion: name your firmware with "portapack" in the title and install into /usr/share/hackrf just like the standard firmware, this gives users one location to search for firmware.

ZeroChaos-, 2015-07-27:
suggestion: use XZ instead of bz2, although for such small files it almost doesn't matter, but on principle XZ is vastly superior

ZeroChaos-, 2015-07-30:
for the scripts, I'm thinking something like this:

#!/bin/sh
dfu-util --device 1fc9:000c --download /usr/share/hackrf/hackrf_one_usb_ram.dfu --reset
hackrf_spiflash -w /usr/share/hackrf/portapack-h1-firmware.bin

#!/bin/sh
dfu-util --device 1fc9:000c --download /usr/share/hackrf/hackrf_one_usb_ram.dfu --reset
hackrf_spiflash -w /usr/share/hackrf/hackrf_one_usb_rom_to_ram.bin

I'm also thinking you need to rename your "hackrf_one_usb_ram.dfu" to something else unless it is the same as the file shipped by hackrf (in which case, stop shipping it at all).  Suggested name would be "portapack-h1_usb_ram_dfu" if it is different.

ZeroChaos-, 2015-07-30:
per discussion on irc, the hackrf_one_usb_ram.dfu is the same as shipped with hackrf and is shipped to make switching to portapack as simple as possible.

jboone, 2015-07-30:
Yep. If hackrf_one_usb_ram.dfu is already present on the system (packaged), feel free to use that copy. In fact, that's preferable to me. Including the .dfu is a shortcut for now, to make sure that you have everything necessary (aside from dfu-util and hackrf_spiflash) when you clone the repo.

jboone, 2015-07-30:
The scripts will work if dfu-util 0.7 is on the system. dfu-util 0.8 has some changes which have broken things. This needs to be addressed, but not right now. Issue #24 tracks that, as well as issue mossmann/hackrf#117.

jboone, 2015-07-30:
I do recall one issue with using the commands as shown in the scripts. The HackRF needs a bit of time to re-enumerate with the operating system after the dfu-util has finished. I would recommend a "sleep 1s" between dfu-util and hackrf_spiflash, to give the HackRF time to show up as a HackRF.

ZeroChaos-, 2015-07-30:
for this second, to make sure that I can push this to users soon, I'm going to leave the scripts with no real error handling and get them into gentoo.  later we can add some ugly checks like "dfu-util -V | head -n1 | awk '{print $2}'" to check dfu-util version and change args if needed, as well as possibly parsing lsusb to determine if the hackrf is even present at runtime.

jboone, 2015-07-30:
Huh, interesting. In my Makefile, I do not specify the product ID, just "--device 1fc9". The product ID in your scripts are correct and work and are more explicit.

jboone, 2015-07-31:
Instructions for entering DFU mode:
"Hold down the HackRF's DFU button (the button closest to the antenna jack) and plug the HackRF into a USB port on your computer. After the HackRF is plugged in, you may release the DFU button."