sharebrained/portapack-hackrf #29: Firmware flashing instructions seem incomplete for hackrf_spiflash step

CaptainMcCrank, 2015-07-26:
I'm following flashing instructions here: https://github.com/sharebrained/portapack-hackrf/wiki/Updating-Firmware

I'm in the firmware restoring section where I'm running into a missing file.  

$hackrf_spiflash -w hackrf_one_usb_rom_to_ram.bin
Error to open file hackrf_one_usb_rom_to_ram.bin

$ls 
LICENSE				portapack-h1-firmware.bin
hackrf_one_usb_ram.dfu

Should hackrf_one_usb_rom_to_ram.bin have been created by a previous step, or is this file something I should have from elsewhere?

CaptainMcCrank, 2015-07-26:
Solution:
Visit https://github.com/mossmann/hackrf/releases/ to download the latest firmware.  
Extract the tar file

Run hackrf_spiflash -w & point to [firmwaredir]/bin/hackrf_one_usb_rom_to_ram.bin

closed 2015-07-26 by CaptainMcCrank

jboone, 2015-07-26:
I will add a note to the Wiki that you need to acquire the binary from the HackRF project, along with a link. Thanks for letting me know.