sharebrained/portapack-hackrf #31: Address mechanical incompatibility with HackRF Blue

jboone, 2015-07-28:


KC5CQW, 2015-08-06:
Mounting holes and/or header socket alignment constancy. 
The socket variable is likely due to solder surface tension pulling the socket mass to equilibrium during reflow process.

jboone, 2015-08-06:
@KC5CQW I'm not sure if you're commenting on a separate issue, or sharing ideas for this issue. This issue relates to the (optional) presence of a header on the HackRF Blue that is not present on the HackRF One. Two electrolytic capacitors on the PortaPack H1 mechanically interfere with the HackRF Blue header. It is not related to the 0.1" SMT headers on the PortaPack, as they utilize alignment pins, and should be positioned to within a few mils of nominal.

KC5CQW, 2015-08-06:
Just sharing ideas... However, based on your input, I can say that by adding three washers in addition to the standoffs cured this problem for me. Enough of the pin length remained to make good contact within the socket headers. With the washers on hand and applied, I have about 0.2-0.4mm (estimated) clearance from the caps.
I own an "official" HackRF but added the P30 header as provided by Nooelec with the RF shield kit.
Sadly, it is not accessible with the H1 mounted. As an option, I suppose a right angled header would cure this. I forgot about the alignment pins... Good call using them in the design!

BTW, i know squat about software but RF hardware test/tune/calibration is my chosen Voo-Doo. 
My wife and I are the PCB assemblers for the AMSAT FOX-1 series cubsats. My hope is to use the HackRF for post assembly smoke testing prior to shipping out the hardware for "real" tests and flight integration.

KC5CQW, 2015-09-05:
When the P30 header is installed (OEM or Blue), an additional 2.35mm of clearance is needed for a total of 14.25mm. I used 15 4-40UNC flat washers (3 on each screw) to provide the additional clearance between PCB's for the caps to clear the header. This setup WILL block access to the P30 header. 
The other option is [for the end user] to source 2 ribbon cables 11 pin (P16) & 13 pin (P22) with one end male , the other female and >1" (25mm) long to allow for an offset mounting configuration between PCB's.