sharebrained/portapack-hackrf #33: Means to adjust receiver channel bandwidth

jboone, 2015-07-29:
Should be easy enough to re-assign the FIR taps.
Adjust spectrum samples decimation factor to make sure aliasing is minimized.

aernan, 2015-08-02:
I could see this being like a zoom in out on the screen feature.

aernan, 2015-08-06:
You can adjust the squelch and receive bandwidth in this app.
android app RF Analyzer https://play.google.com/store/apps/details?id=com.mantz_it.rfanalyzer
probably do something similar.

jboone, 2016-01-05:
Plumbing is there as of commit 28be223ace91d6bac55dc468789ffa0de57904c4. (It took until this commit to get something that was compilable, due to a merge-oops.)

jboone, 2016-02-06:
NBFM channel filter can be varied between 8.5kHz, 11kHz, and 16kHz widths, with attendant adjustments to deviation (2.5 or 5k), as of commit ceef8d32d3b5da9a4faf7b98cad7ae677c63a600.

KC5CQW, 2016-05-05:
Please explain the deviation adjustments. Is this automatic or user variable?

jboone, 2016-07-05:
Closing this and continuing work on channel bandwidth adjustment for CW/AM/SSB via issue #86.

closed 2016-07-05 by jboone