sharebrained/portapack-hackrf #35: Provide checkmark to disable receiver NFM squelch

jboone, 2015-07-29:
Sometimes, you just want to open it up and listen. I'm thinking satellite work in particular, where signals are very weak initially, and hearing *anything* is a help to aiming an antenna.

KC5CQW, 2015-11-23:
I tried to raise the gain to override the squelch, this did not work very well as it overloaded the front end.

KC5CQW, 2016-05-05:
What about adding a top line selection (0-10 level range) up there with the screen shot and sleep icons?

jboone, 2016-05-05:
@KC5CQW Raising the gain in FM would not change the squelch unless it also changed the SNR of the signal -- which is possible, but not predictable.

The squelch value exists in code, it simply needs to be exposed as a control and stored somewhere persistent.

I'd like to keep the top line free of app-specific features, as those icons represent features available globally. I'm sure we can find another place to shoehorn the squelch, where it's reasonably accessible. Perhaps it's an option that appears when the audio level is focused.

jboone, 2016-05-05:
Oh, and an audio options view might also contain a control for the wideband FM equalization mode (75us vs 50us de-emphasis). Probably other things, too...