sharebrained/portapack-hackrf #42: HackRF firmware image padding with dd yields zero-byte file.

jboone, 2015-08-02:


jboone, 2015-08-25:
Reports from issue #55, from a Ubuntu 15.04 user, are that this fix isn't working. Reopening...

splaspood, 2015-08-25:
head -c 0x10000 /dev/zero| cat hackrf_one_usb_ram.dfu - | head -c 0x10000 > hackrf_one_usb_ram_dfu_pad.bin
head: 0x10000: invalid number of bytes
head: 0x10000: invalid number of bytes

splaspood, 2015-08-25:
Ok, I think I see the problem.   I had a 0 byte file lying around from before, and make clean doesn't remove it, so when I rebuilt it did not re-create the pad file using the new head method.    Rebuilding now to confirm.

splaspood, 2015-08-25:
Confirmed.

jboone, 2015-08-25:
I should rewrite the image-building process as a Python script. It will be more predictable. And as a side benefit, it should work in Windows, unlike my current cat and head and /dev/zero approach, or even the older dd approach.

jboone, 2015-08-25:
Rewritten and committed. I'm closing this again. Hopefully I won't have to open it a third time. :-)

commit 255de16d (2015-08-02): Use head/cat, /dev/zero to pad HackRF firmware

dd approach didn't work for at least one user. Resolves #42.
closed 2015-08-25 by jboone