sharebrained/portapack-hackrf #46: Portapack loses waterfall display or locks up.

topcyde, 2015-08-03:
Have issues with portapack waterfall display stopping or device locking up with commit 255de16d3eff1baf3c806c280153c8d5df0e112f.  Have not tested against earlier commits, but did test against precompiled binaries.
    Usually happens on initial plugin, but operates fine until going into the reciever mode.  I can consistently 2 out of 3 times get it to do it by pressing reset shortly and then immediatly going into the receiver mode.  It will scan for anywahere from 2 seconds to about 15 seconds before issue occurs.  Cannot get this error to repeat using the precompiled binaries.

jboone, 2015-08-03:
Which modulation mode are you using when the issue occurs? (AM, NFM, WFM, FSK?)

topcyde, 2015-08-04:
It will do it in any mode, assuming I can scroll over to it and change it before it locks up.  But usually it's in NFM, but only because it's the default.  I did do some testing last night and if it doesn't lockup within the first 18 seconds it will run for hours without ever erroring.

jboone, 2015-08-04:
It appears, from further reports, that this is related to the build environment somehow. Binaries of a more recent commit, produced on a different system, appear to run fine.

topcyde, 2015-08-05:
Problem was source being built using version 4.8.4-1+11-1.  Compared to same commit compiled on a different system, showed now signs of error.  Updated to version 4.9.2-10+14-bpo8+1.  Recompiled, new bin file shows no signs of previous erroneous behavior.

closed 2015-08-05 by topcyde