sharebrained/portapack-hackrf #47: switch_to_portapack dfu-util output

ZeroChaos-, 2015-08-03:
gato ~ # switch_to_portapack
Hold down the HackRF's DFU button (the button closest to the antenna jack)
then plug the HackRF into a USB port on your computer.
After the HackRF is plugged in, you may release the DFU button.
Press any key to continue or ^c to abort

dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1fc9 product = 0x000c
Opening DFU capable USB device... ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
WARNING: Runtime device already in DFU state ?!?
Found Runtime: [1fc9:000c] devnum=0, cfg=1, intf=0, alt=0, name="DFU"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
DFU CRC does not match
Warning: File has no DFU suffix
bytes_per_hash=409
Copying data from PC to DFU device
Starting download: [##################################################] finished!
unable to read DFU status
can't detach
Resetting USB to switch back to runtime mode
error resetting after download
File size 207564 bytes.
Erasing SPI flash.
Writing 207564 bytes at 0x000000.

This is what it looks like when I run the switch_to_portapack script.
I really don't like this bit:
unable to read DFU status
can't detach
Resetting USB to switch back to runtime mode
error resetting after download

That said, both scripts seem to work flawlessly anyway.  Is this normal or something we need to consider improving?

jboone, 2015-08-03:
It's normal *and* something we need to improve. It's noise from dfu-util that doesn't actually impact the process. Also, the barfing of hundreds of "Writing XXXXXX bytes at 0xYYYYYYY" is something that will be silenced by default when my HackRF patch is accepted.

ZeroChaos-, 2015-08-03:
Would you perhaps like to silence the dfu-util output in the scripts?  I can run some tests silencing stderr or everything and make it less ugly by just reading error codes possibly.

jboone, 2015-08-04:
@ZeroChaos- Please do. I don't see any command line options for dfu-util to silence the noise, so it will no doubt require some dirty tricks.