sharebrained/portapack-hackrf #48: Additional Demodulation Modes

KC5CQW, 2015-08-05:
Implement Dubble side-band, Upper and Lower side-band, Contentious Wave modes for use in spectrum analysis and communications monitoring. 
Provide variable pass-band tuning width and carrier offset features to accompany all demodulation types. (This may be a separate "issue" request, sorry!)

KC5CQW, 2015-08-06:
Yes, similar to #33.

KC5CQW, 2015-11-23:
If possible, I/Q output on the audio line for demodulation by a secondary "computer" would be great as well. This can be a simple settings check box. Demodulated audio vs. raw I/Q.

jboone, 2016-02-06:
AM (DSB) and SSB (USB/LSB) are supported as of commit ceef8d32d3b5da9a4faf7b98cad7ae677c63a600.

KC5CQW, 2016-02-06:
Yes!, this makes me happy!
Now for a future feature option request... Auto display zoom when changing BW. Oh yea, and TX of course. 

73, Damon

closed 2016-02-15 by KC5CQW