sharebrained/portapack-hackrf #50: Hardware: alternative case proposal

dc1rdb, 2015-08-07:
Alternative case, based on the original HackRF plastic enclosure. The bottom half of the case could be reused. Just the top half of the plastic enclosure would need to be replaced by a version with extended side walls and cut-outs at the top for display and navigation control, plus additional hole for headphone jack and slot for SD-card.

jboone, 2015-08-07:
@dc1rdb That's a great idea. However, I'm not much of a fan of plastic cases, and the cost of getting injection molding set up is vastly higher than I can afford.

But with 3D printers everywhere, perhaps a printable model could be made, based on your idea. If somebody wants to tackle that before I get to it, I highly recommend both OpenSCAD and FreeCAD. I have OpenSCAD models of the PortaPack and HackRF boards in the repo already that could be used to validate the positions of holes, edges, etc. I'm not sure if I posted my FreeCAD files yet...

KC5CQW, 2015-08-07:
Sadly, the nature of this hardware [almost certainly] requires a metallic case for EMI mitigation. Such is the nature of RF.
It would be possible to print a case then line it with foil tape as an alternative with "limited" success.

For those not critical of EMI, or as a stop-gap measure, a printable case is a great idea to protect this valuable hardware. Personally, I would be willing to act as a beta tester and print one out if provided an .stl file. I am unable to commit to drafting a model at this time due to other obligations.

ne1, 2015-08-22:
I have done a case, but i have to redesign it a little bit, to make it fits better. I will upload it on monday. 
I can send these to somebody, if anyone needs it.

ne1, 2015-08-25:
Hi, sorry it takes a little longer and there is still room to improve this model, here is one of my test cases:

[image]
[image]
[image]

You can get it here: 
http://www.thingiverse.com/thing:983995

dc1rdb, 2015-08-26:
This is looking great! Regarding KC5CQWs concerns about RF shielding, this could be addressed by covering the inside with conductive graphite spray paint.

ne1, 2015-08-26:
Thanks! If anybody interested: i can print and send it for just the shipping costs :) (sitting in Germany) 

This spray should not touch the bottom side of the hackrf, correct? This could be possible in the actual design. 
But i can give a try. Is aluminium foil a suitable alternative?

KC5CQW, 2015-09-05:
Add 2.35mm between the HackRF's button and SMA holes and the PortaPack's face to the case design to allow for 14.25mm spacers when a HackRF Blue or HackRF with the P30 header is installed. Generate a second set of STL's for this optional configuration.

ablanquart, 2016-01-24:
If enough people are interested, I may be able to have this made in metal. I'll see if I can run a "sample" using ne1's thingiverse file.

ne1, 2016-01-24:
I can send you my FreeCad File if you like!?

ablanquart, 2016-01-26:
Absolutely, please send! Not sure if you can convert or make easily - but a
.dxf or a .dwg file would be better.  I think I can work with the STL and
FreeCAD files though.

ne1, 2016-01-26:
Hi,

here is the file - i hope it can help.