sharebrained/portapack-hackrf #52: add step size for WFM 0.3 MHz

aernan, 2015-08-07:
FM radio stations are 0.3 MHz apart. Please add a new step size when in WFM mode.

KC5CQW, 2015-08-07:
But!, keep the other step sizes as well.
WFM is also used for other modes like NOAA WxFAX reception and it requires Doppler correction to keep clean image clean. In fact, all modes would benefit from 10M, 5M, 1M, 100k, 10k, 9k, 3k, 1k, 100H, 10H and 1H (for drift measurements when a clock standard is used on PLL1).

KC5CQW, 2015-08-07:
I ment "Hz" not "H(enry's)".
...cps is what I prefer over Hz myself...

jboone, 2015-08-29:
You should note that the step size is a completely independent parameter from the demodulation mode. So you can use 10MHz steps with AM or NBFM, and 100Hz with WFM, and any other combination thereof.

My experience and all information I can find suggests that broadcast FM is spaced at multiples of 200kHz, and either even or odd tenths, depending on the country. See [Wikipedia](https://en.wikipedia.org/wiki/FM_broadcast_band) for more. Seems like the 100kHz mode should work well for this. If the concern is fast tuning between stations, I would recommend instead adding a memory/preset/favorite feature and tuning mode. Mark the FM stations, and then flit between them with a single click.

I'm interested in the justification for 9k and 3k. I definitely appreciate the Doppler correction, though I suggest checking out the PPM adjustment in the latest firmware. If PPM and Doppler are separate, I'd suggest having a separate Doppler adjustment. That way, you can tune to the desired center, corrected by the PPM value, and then tune up to where the anticipated Doppler shift will put the signal. In fact, Doppler could even be performed automatically, or in a guided fashion, eventually...

Regarding drift measurement, is this to characterize the internal crystal vs. an clock standard, or to characterize an RF signal against the clock standard?

KC5CQW, 2015-08-29:
Justification for 9k & 3k; same issue with WBFM 200kHz... bandwidth dependent tunning. 9kHz for AM broadcasts and 3kHz for SSB signals. Probably not worth the effort, just a thought as these ranges are avalible in other portable (consumer) RF devices. Being that these ranges are BW dependent by convention, it would be much more beneficial to focus on resolution instead. How low can we go? 10's of Hz? 100's of Hz?

PPM and Doppler are indeed separate! 
As an example, Orbitron is a good program (among many others) for satellite prediction/tracking that calculates the Doppler shift and outputs serial data for tuning in real time. 

PPM drift in the LO clock translates to internal stability of the hardware. Keep in mind that small amounts of drift can easily translate to 10's of kHz offset in the GHz region. 
An external standard such as a GPSDO of 10MHz on the PLL1 input or a sync to a local GSM carrier (also timed via GPSDO at the tower site) can be utilized. 
The nessesary program would characterize the internal clock drift by comparing it with a clock standard as the internal temperature varied. Setup would require the end user to perform a calibration procedure utilizing a means of thermal "ramping" over the expected environmental temp range so the internal temp sensor data and clock signal(s) data can be loaded into a lookup table. For instance, 60-90 Deg. F (or whatever) for inside/outside excursions. This would provide a better stability than no clock standard or raw hardware clock signal alone. Again, the drift in the GHz range can quickly become troublesome with just a few hz offset in the LO clock. This feature would provide a meaningful enhancement for the portable user in times where a reference clock is not feasible and thermal variance is an environmental fact that impacts clock xtal stability.

jboone, 2016-01-28:
The requested smaller steps (3k, 5k, 6.25k, 9k) were added in a recent commit and will be in the next firmware release. I had a harder time justifying the large steps (300k and 5M).

closed 2016-01-28 by jboone