sharebrained/portapack-hackrf #55: HackRF mode broken

jboone, 2015-08-17:
Between binary shared with a few about a week ago, and building from scratch (newer commits), it stopped working...

jboone, 2015-08-17:
A user tracked the problem to consecutive commits 66320c39d4fa62bd322c85734ab62fc627457caa or 4e0de9c4ad04758b5bd4393482e91996ad939186, which makes some sense. Clock configuration of the M4 is being frobbed differently starting there.

Looks like I have more shutdown work to do to provide the correct state for HackRF firmware to run. Chances are, the M0 code needs to reset or halt the M4 and change its clock source back to IRC.

jboone, 2015-08-17:
It appears to be the SysTick configuration in the M4's hal_lld_init(). Chances are, the SysTick interrupt is firing while I'm copying HackRF firmware over the top of the RAM where the SysTick interrupt code *was*. I should shut down the M4 more cooperatively -- probably by adding a M4->M0 message to ask it to quit, and waiting for an M4->M0 message confirming the mode change.

jboone, 2015-08-24:
Should be resolved in commit 11561659b4479636c62127fc477df82f5b3cf8e3.

splaspood, 2015-08-25:
I still seem to have a problem with this commit.    The code appears to build fine, I flash the device, but upon reset the LCD remains black.   I'll gladly provide whatever troubleshooting information you might need if you direct me in retrieving it for you.

jboone, 2015-08-25:
The LCD will remain black, as all PortaPack-related features are disabled when entering HackRF mode. Are you able to access your HackRF from the host computer, e.g. the hackrf_info command?

splaspood, 2015-08-25:
Maybe I misunderstood how HackRF mode is supposed to be enabled, I thought there was a menu option to enable/disable.

Currently, however, after flashing with the LCD dark (and upon a fresh USB connection) hackrf_info does not see the device.   It actually appears that my system does not see the device at all.  (lsusb does not show anything)

jboone, 2015-08-25:
OK, I think I understand. Are you using the latest commit on master (presently 8f326e2d8e4d515e0baee8d95d07f5fed0239b1b)? What is your build operating system and ARM GCC compiler version? And after building a binary, do you have other files in the firmware/ directory with these file sizes:

     20484 hackrf_one_usb_ram.dfu
     65536 hackrf_one_usb_ram_dfu_pad.bin

...and a binary with this *approximate* file size?

    342264 portapack-h1-firmware.bin

splaspood, 2015-08-25:
Yup, latest commit.   I can go back and compile faad4ea053b611cffced165865c873bee34e3255 with this setup seemingly fine.

gcc version 4.8.3 (4.8.3-18ubuntu2+12) on an Ubuntu 15.04 host  (installed from the distribution provided packages)

Regarding the after-build files.   The pad.bin is 0 bytes (certainly my problem) and the portapack-h1-firmware.bin is 276244 bytes.

-jwb

> On Aug 25, 2015, at 2:37 PM, Jared Boone <notifications@github.com> wrote:
> 
> OK, I think I understand. Are you using the latest commit on master (presently 8f326e2)? What is your build operating system and ARM GCC compiler version? And after building a binary, do you have other files in the firmware/ directory with these file sizes:
> 
>  20484 hackrf_one_usb_ram.dfu
>  65536 hackrf_one_usb_ram_dfu_pad.bin
> 
> ...and a binary with this approximate file size?
> 
> 342264 portapack-h1-firmware.bin

jboone, 2015-08-25:
Yep, the zero-byte file is definitely the problem. It will misalign the "application" image that runs on the M0 and runs the user interface. Earlier, I was using a dd trick to pad the HackRF firmware to 64K, and had this exact problem (see issue #42). So I changed approaches and thought I'd resolved it. Maybe I should just write an image-building script in Python and not try to shell my way out of the problem.

splaspood, 2015-08-25:
Bingo.   I manually used dd to create the pad as was done previously, then 'built' the image again.   Looks to be working properly now.

jboone, 2015-08-25:
Reclosing. The problem @splaspood was having were caused by a derivative of issue #42.

jboone, 2015-09-01:
Reopened. Sigh. However, I bisected and identified commit 7937ea7327a461646fb2b0808bd3ebed7f65d2e5 as the culprit. What's troubling me is I have NO idea why. But if I remove that "__DMB();" line on HEAD, HackRF mode works just fine. WTF?

closed 2015-09-01 by jboone