sharebrained/portapack-hackrf #60: OS X compiler installation issue.

jboone, 2015-08-26:


jboone, 2015-08-26:
@KC5CQW reports "I couldn't get the ARM compiler to instal on my X.9 system."

jboone, 2015-08-26:
@KC5CQW What package manager are you using to install the ARM compiler -- MacPorts or Homebrew or other? Or are you downloading the GCC-ARM toolchain binary from Launchpad? The more details you can provide, the more I can help.

KC5CQW, 2015-08-29:
Sorry, I was attempting to install the GCC-ARM tool chain but now my "free time" is occupied with school work. I will be monitoring the progress for this but not able to actively participate on this issue for now. Basically, the procedures don't jive with the actual software; couldn't install. I'll try again soon and clarify.

0ff, 2015-08-31:
For the record, I have no problems at all on OS X 10.10.5 using `gcc-arm-none-eabi-49` (https://github.com/px4/homebrew-px4/blob/master/gcc-arm-none-eabi-49.rb) installed via homebrew.

jboone, 2016-04-20:
Closing, as I've heard from many people who have had success -- and the ARM GCC compiler has iterated a few more releases since (now GCC 5.3).

closed 2016-04-20 by jboone