sharebrained/portapack-hackrf #63: WFM waterfall

dc1rdb, 2015-09-05:
The list of fixed bugs for firmware 20150904 includes an item "WFM waterfall freezes". However, WFM waterfall does not seem to be implemented in that firmware version.

KC5CQW, 2015-09-05:
Omission could be the current "fix"(?).

jboone, 2016-01-05:
Fixed in commit dcb2c46c57841a010be07cc85d54a09ef2baf830.

closed 2016-01-05 by jboone

KC5CQW, 2016-01-05:
This is the message when I try to download new FW:

PermanentRedirectThe bucket you are attempting to access must be addressed using the specified endpoint. Please send all future requests to this endpoint.portapack-h1-buildsportapack-h1-builds.s3.amazonaws.com75DF1398594921207J3pwvbYxhm6kEYq5BLZVkkQ/s4SxwaafsvicmggcnhWY1lLWiFr7GC1yOCXL3tznh2ruqxSzUU=

jboone, 2016-01-05:
Argh, yes, you have to modify the URL to change the "s3.amazonaws.com/portapack-h1-builds" bit to "portapack-h1-builds.s3.amazonaws.com". For example, use: https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/42/42.1/firmware/portapack-h1-firmware-dcb2c46.zip