sharebrained/portapack-hackrf #65: Add audio codec 8kHz mode for less demanding audio modes.

jboone, 2015-09-10:
No point in doing DSP at 48kHz when your demodulated signal has 3-4kHz of bandwidth...

Whether the codec can support other rates other than 8/44.1/48/96 should also be investigated. I don't see any reason why the codec would care.

KC5CQW, 2015-09-11:
Usually, 11.6kHz is just fine and allows a little headroom for any FEC, good SNR, IMD, AFC drift/tracking etc. Higher sample rates are good for multiple signal decoding but not feasible on the H1 due to LCD (and CPU) real estate. 8kHz should be just fine for simple demodulation such as CW, RTTY etc.

jboone, 2016-02-06:
Support is complete in the new audio API, as of commit 029416548159c7dfb110b120a1298782332e413c. AM/SSB, NBFM modes take advantage of lower audio sampling rates.

closed 2016-02-06 by jboone