sharebrained/portapack-hackrf #67: Changing ppm offset in main menu doesn't take effect until changed on waterfall

r4v5, 2015-10-12:
Steps to reproduce:
* Power on the Portapack
* Wheel down to Setup -> Frequency Correction
* Change this value
* Back out to the main menu, start Receiver
* Observe that the waterfall hasn't changed to reflect the offset, but if you highlight the frequency it lists the new PPM value
* Select PPM value, rock left or right
* PPM now immediately jumps into effect

This happens on a portapack running aa733b1. The offset does get stored and persists across power-on/power-off, though, which is nice.

jboone, 2016-01-28:
Hmmm, I think I was able to reproduce this. Notes to myself for when I tackle this: PPM seems synchronized between receiver view and setup view. But actual tuned frequency is not altered after changing PPM in setup view -- returning to receiver view, signal still appears at offset seen before the PPM adjustment.

jboone, 2016-02-06:
I believe this is addressed in commit a9451c17b4ee18ddb7755b9111990a0b25ddeda9.

closed 2016-02-06 by jboone