sharebrained/portapack-hackrf #68: unable to compile firmware ../common/lpc43xx_cpp.hpp:25:19: fatal error: cstdint: No such file or directory

mgdudas, 2015-10-30:
I'm unable to build the current firmware.

I get the following error:

In file included from main.cpp:25:0:
../common/lpc43xx_cpp.hpp:25:19: fatal error: cstdint: No such file or directory
 #include <cstdint>
                   ^
compilation terminated.
../chibios/os/ports/GCC/ARMCMx/rules.mk:165: recipe for target 'build/obj/main.o' failed
make[1]: *** [build/obj/main.o] Error 1
Makefile:69: recipe for target 'baseband/build/baseband.elf' failed
make: *** [baseband/build/baseband.elf] Error 2

This is after updating to the most recent src tree via git pull. I have also tried deleting everything and doing a new git fetch. I see the same problem.

Building on Ubuntu 15.10 AMD64 in case it matters.

jboone, 2015-11-02:
I'm curious how you installed your ARM GCC compiler. "cstdint" is a standard C++ header, so your error suggests to me not having the G++ portion of the GCC compiler suite installed.

mgdudas, 2015-11-02:
Hmm, well, I had installed everything on this machine quite some time ago, maybe at least 2 years from when I got my hackrf originally. The only thing I can think of would be that I updated to Ubuntu 15.10 since the last time I successfully built the firmware. So maybe something got screwed up there. In fact, it looks like https://launchpad.net/gcc-arm-embedded hasn't released updated tools for ubuntu 15.10, so that may be it. I'll try on another machine with a clean install. Thanks!

jboone, 2015-11-08:
After another report of this problem, I updated my Ubuntu to 15.10 and was able to reproduce with the simplest possible example, a single C++ source file: https://gist.github.com/jboone/8a7abbba27df442f011e

jboone, 2015-11-08:
Attempting to include other C++ standard header files (e.g. cstddef, memory, algorithm, string) yields the same error. Simplifying the compiler command line to remove -mcpu, -mthumb, --specs have no effect on the error.

jboone, 2015-11-08:
Further investigation compiling with "-v" option reveals that there's some mis-named directories in the Ubuntu package:

```text
gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn227297-1) 
COLLECT_GCC_OPTIONS='-mcpu=cortex-m4' '-mthumb' '-specs=nano.specs' '-std=c++11' '-v' '-o' 'test_cstdint.elf'
 /usr/lib/gcc/arm-none-eabi/4.9.3/cc1plus -quiet -v -imultilib armv7e-m -D__USES_INITFINI__ test_cstdint.cpp -quiet -dumpbase test_cstdint.cpp -mcpu=cortex-m4 -mthumb -auxbase test_cstdint -std=c++11 -version -o /tmp/ccFrI9pJ.s
GNU C++ (15:4.9.3+svn227297-1) version 4.9.3 20150529 (prerelease) (arm-none-eabi)
	compiled by GNU C version 5.2.1 20151003, GMP version 6.0.0, MPFR version 3.1.3, MPC version 1.0.3
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/include/c++/4.9.3"
ignoring nonexistent directory "/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/include/c++/4.9.3/arm-none-eabi/armv7e-m"
ignoring nonexistent directory "/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/include/c++/4.9.3/backward"
ignoring nonexistent directory "/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/sys-include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/arm-none-eabi/4.9.3/include
 /usr/lib/gcc/arm-none-eabi/4.9.3/include-fixed
 /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/include
End of search list.
```

If I perform these steps, it appears to address the issue:

```bash
cd /usr/lib/arm-none-eabi/include/c++
sudo ln -s 4.9 4.9.3
```

It seems to me this is an Ubuntu or GCC bug.

closed 2015-11-08 by jboone

mgdudas, 2015-11-09:
Hi,

Yep, that fixed it. Thanks for your help!

havronl, 2019-03-23:
I now have the same problem... Looks like versioning issues.  I am not a skilled compiler.  Can some give me the command to run compiler with -v option so I can see what needs to be linked?