sharebrained/portapack-hackrf #69: Spec. Mode Waterfall Speed

KC5CQW, 2015-11-23:
The waterfall scroll speed is too fast for manually tuning onto a signal once the transmission ends. This was not an issue in the September release.

jboone, 2015-11-23:
Yes, this is a behavior in unreleased firmware. The waterfall will return to normal scroll rate once I can reclaim enough code (.text) section RAM to add back the window-presum-FFT window to the spectrum calculations.

Longer term, the scroll rate, averaging, etc. will be controllable. But that's tough to achieve right now in the 32K of available baseband processor RAM...

KC5CQW, 2015-11-23:
Ahh, thanks for the explanation!

jboone, 2016-01-05:
Speed is addressed in current (but unreleased) firmware. Update rate is now ~75Hz, which might still be a bit faster than ideal. I should open a new issue about adding configurable spectrum scroll rate.

closed 2016-01-05 by jboone