sharebrained/portapack-hackrf #70: TPMS & AIS Frequencies

KC5CQW, 2015-11-23:
When selecting the TPMS or AIS demodulator modes, automatically set the frequency to the most likely (expected) value. This is to expedite reception by tuning the used frequency per the selected mode for us "lazy folks" that do not know the frequencies by heart. 

 Examples:  marine AIS is performed on 87B (161.975 MHz) and/or 88B (162.025 MHz) GMSK. Some method of locking onto the active channel or manual selection of CH87 or CH88 will be needed once the AIS mode is selected. 

 315MHz (or 434 for Europe) for TPMS may be "ball parked" and then manually fine tuned by the user to account for TX variances. (Unless they are generally right on 315MHz or the RX bandwidth and demodulator can allow for some spread).

jboone, 2015-11-23:
The TPMS and AIS features are still half-baked. Maybe more like quarter-baked. Hence they're not yet in an official release. Limiting the tuning is definitely in the cards, though my preference would be to break these functions into entirely separate "applications", perhaps accessible under another top-level menu (not "Receiver"). Maybe "Digital Modes" (sounds too much like PSK31, APRS, WSPR, etc., though...) or something transportation-themed (maybe "Transponders"?). This would permit more extensive message decoding than the Receiver UI would comfortably allow. Something more like Kismet, where the N most recent transmissions are kept in a list, and details about each can be viewed full-screen...

jboone, 2015-11-23:
...and another justification for a separate application. For AIS, I think it's quite possible to decode both channels simultaneously, so no tuning (other than fine/PPM) required. For TPMS, I would like to decode as many different modulations and formats as I can. Whether that extends to transmissions at substantially different frequencies within the band will depend on how many CPU cycles I have remaining. If I run out, the user will need to choose a center frequency within the band, and will miss weaker messages significantly off-center (due to loss of sensitivity outside the matched filter's bandwidth).

KC5CQW, 2015-11-23:
I agree with the "transponder" category, good call. Digital is a bit generic of a term and would be best used for the 2-way modes that you've listed.

jboone, 2016-01-28:
There's now a transponder menu, and the frequency selection is simplified, only adjustable to the extent necessary. In the case of AIS, the two VHF channels you mentioned are the only options. For TPMS, there's no adjustment, since so far I only support transponders I've found on one frequency (315MHz). For ERT, I'm still deciding what makes sense, and will incorporate before release. However, ERT's algorithm does wideband processing, and receives transponders over a range of a couple of MHz. So it may not be a center frequency adjustment, so much as a range adjustment.

closed 2016-01-28 by jboone

dc1rdb, 2016-01-28:
Please add frequency selection support for European TPMS at 434 MHz. Thanks!

jboone, 2016-01-28:
@dc1rdb That's definitely the plan! I will need time to learn if there distinct TPMS modulations and packet encodings in other regions of the world, then I will implement support for those. At the moment, I am implementing what I have access to.

dc1rdb, 2016-01-29:
FYI: the European system works at 433.92 MHz and is described in great detail at http://elib.dlr.de/81155/1/TPMS_for_Trafffic_Management_purposes.pdf
Hope this helps!

jboone, 2016-01-29:
Thanks, I appreciate the help.

That paper reiterates what I've known about TPMS technology for a while now -- it's not a single system or protocol, but a bunch of different, slightly different devices, each of which needs a certain amount of custom handling to correctly interpret. The exact center frequency, modulation characteristics, symbol encoding, CRC or checksum, all vary. Schrader, one of the dominant TPMS vendors has claimed -- as of 2009, I think -- that there are 140 TPMS variations. Schrader is quick to point out that they offer a sensor that can be configured to speak most of those variations, once you've configured it based on the year/make/model of your vehicle and the variant that car's ECU speaks. It is possible that the TPMS support I've implemented *may* work for a few European models at 433.92MHz, but I have no way of validating that.

To some extent, I think this variety was by design, so they didn't have to take the time to work together or create a single device ID assignment authority.

In short, to support European TPMS devices will require direct research, analysis, and reverse-engineering of signals from many different vehicles in that region.