sharebrained/portapack-hackrf #71: ADS-B Decoding

KC5CQW, 2015-11-23:
Add a demodulation mode for ADS-B aircraft transponders used on 1090MHz.

jboone, 2015-11-23:
My recent work on OOK TPMS demodulation suggests that ADS-B may be feasible after all. The only challenge will be symbol clock recovery, which right now is not at all optimized, and may not be able to keep up with the ~4MHz sample rate it requires. But there's much room for improvement in that code, which is implemented in floating point, at present.

KC5CQW, 2015-11-23:
See my last comment on #48. The I/Q data output would allow for off-board decoding in the event CPU/RAM overhead is maxed out. This could be a temporary solution until code headroom is available. Either way, I find it intriguing that such complex decoding can/may/will be done on such a portable, low power device!

jboone, 2015-11-23:
It would allow off-board decoding for low-bandwidth signals. ADS-B/Mode S signals are several MHz wide. The WM8731 audio codec is only capable of 96kHz sampling rate, which would permit only 96kHz  of complex spectrum to leave the device through the audio output. To look at it another way, pulse widths in aircraft RADAR systems are approximately 0.5us long. A single audio sample from a 96kHz audio codec is 10us long -- 20 times the duration. Even if you converted baseband to amplitude, there would be no way to represent such short pulses via the audio codec. Some amount of demodulation or packet detection and buffering would have to occur if audio output is used for data egress. I would recommend other options, like using the USB interface. But at that point, we're reinventing the HackRF firmware and the Mode1090 project.

jboone, 2015-11-24:
For the future, here's some references I've found for Mode A/C/S and ADS-B:

http://tech-software.net/radar/
http://www.airsport-corp.com/modec.htm

KC5CQW, 2016-05-05:
Fair enough, I keep forgetting that loading the HackRF mode IS for off-board reception (derp).

KC5CQW, 2019-03-26:
Now handled with Furtek.

closed 2019-03-26 by KC5CQW