sharebrained/portapack-hackrf #72: Push to Talk/Voice Operated Switch for RF transmissions

KC5CQW, 2015-11-23:
All regulatory agency and local laws must be observed!

Add a red PTT icon on the RX screen so that when the curser has it highlighted, the center button of the thumb wheel may be used as a PTT switch to allow for RF transmission. 
Employ a VOX system for "hands free" transmission capability when sufficient audio/data levels are able to trigger transmission. (Adjustable to typical levels)1v p-p on the mic line)).
Provide a PTT lock-out in the settings to prevent unintentional transmission (have this enabled by default and at power up, reset etc. so the user has to "opt-in"). 
Provide a transmit time out timer with adjustable time ranges (incremental by 30 second steps for example) to prevent a stuck or unintended transmission.

jboone, 2015-11-23:
I definitely intend to present a warning before any TX operation, which must be acknowledged before TX can occur. The warning would be presented once per power-up, on first TX. I think we're in agreement. :-)

As for a TX timeout, it would apply only to certain modulation schemes (primarily audio -- AM/SSB/NBFM). Packet-based TX wouldn't need it.

KC5CQW, 2015-11-23:
Agreed, however... A backup timer for just in case things get stuck in data TX may not be a bad idea. But then again... If a lock-up occurred, the timer would likely not work anyway. Hmm, I guess some kind of watch dog timer at a lower level would be needed.

jboone, 2015-11-23:
The LPC4320 does have a hardware watchdog. It could be pressed into service to mind the entire system, not just TX.