sharebrained/portapack-hackrf #73: Travis-CI build firmware URLs are wrong.

jboone, 2016-01-30:
Correct URL involves replacing "s3.amazonaws.com/portapack-h1-builds" with "portapack-h1-builds.s3.amazonaws.com".

The problem is deep within the Travis-CI "Artifacts" magic box. I've tried and failed to set the correct hostname for those URLs.

miek, 2016-01-30:
Just for reference: the original fork of goamz has an old PR that seems related https://github.com/goamz/goamz/pull/33

dominicgs, 2016-01-30:
A quick hack to fix this may be adding an echo here: https://github.com/sharebrained/portapack-hackrf/blob/master/.travis.yml#L19
Something like this:
`echo "Attempting to upload to https://blah/blah/blah"`

dominicgs, 2016-01-30:
Oh, I see my error there.  Ignore that comment.

KC5CQW, 2016-02-03:
That which has been seen, cannot be unseen.

dc1rdb, 2016-02-05:
Latest Travis-CI log does not contain download URLs at all.

jboone, 2016-02-05:
They're there in the latest log (90 minutes ago), but are still incorrect as described above. I've given up on solving this for now.

dc1rdb, 2016-02-05:
Thanks - works! I don't mind modifying the URL, as described above.

jboone, 2016-02-05:
@dc1rdb Fair enough! It's just an unsightly and user-inconvenient thing that gnaws at my brain periodically. I will address it, but it will take more work than I want to allocate to it right now. Thanks for being flexible. :-)

dc1rdb, 2016-02-05:
You're welcome! Thanks for working so hard on the actual firmware, appreciate it a lot!

jboone, 2016-06-27:
A PortaPack user and good neighbor set up a redirector at [http://munchkins.net/portapack](http://munchkins.net/portapack). Use that URL as a quick way to get the latest PortaPack build, while this issue is still unresolved.

jboone, 2016-07-10:
Whoops, the "munchkins" redirector, mentioned in the prior comment, is now broken due to the firmware directory now being inside a "builds" directory, as is CMake's way.

aSmig, 2016-07-10:
The http://munchkins.net/portapack redirect should be working again.  Note that the travis-ci download URL posted on IRC was previously correct and now requires insertion of build/ before firmware/.

So, present best methods for downloading the CI build:

1. http://munchkins.net/portapack (easy!)

2. https://travis-ci.org/sharebrained/portapack-hackrf/
2.b. Scroll to bottom and find "download_url:"
2.c. s/s3.amazonaws.com\/portapack-h1-builds/portapack-h1-builds.s3.amazonaws.com/

3. irc://freenode.net/#portapack
3.b. Watch for <travis-ci> to announce "Firmware download"
3.c. s/firmware\//build\/firmware\//

jboone, 2016-07-10:
@aSmig Whoops, as discussed privately, I have an error now in my .travis.yml. Once I fix it, the URLs you're using will be accurate again. Do I need to synchronize with you to commit that change, so that Munchkins will still work?

jboone, 2016-07-10:
OK, Travis-CI URLs in Freenode #portapack IRC, and redirection via munchkins.net are now correct. Sorry for the inconvenience...

jboone, 2020-02-05:
CI builds have been redirected here: https://jboone.github.io/portapack-hackrf-nightly/

Closing.

closed 2020-02-05 by jboone