sharebrained/portapack-hackrf #75: Bandwidth adjustment remains after adjusting from NFM to WFM or SPEC.

rjmendez, 2016-02-13:
Minor UI glitch noticed after flashing 937dad6, the bandwidth adjustment will remain visible when scrolling through modes. Possibly added in 3f86a7c?

jboone, 2016-02-13:
Thanks. I think it's a repaint problem. The bandwidth options view can't be navigated to (receive focus), so it suggests it's not really there. I'll check the code and see what I've missed.

jboone, 2016-02-13:
Ugh. This is due to a terrible weakness and lack of proper design effort of the widget painting scheme. I don't think I'll have time to fix it (properly) before the next release. There might be a dirty, filthy, disgusting work-around I could manage...

jboone, 2016-07-05:
This issue is addressed in some indeterminate commit in the last month or so. It will appear in the next release, and appears now in Travis-CI builds.

closed 2016-07-05 by jboone