sharebrained/portapack-hackrf #76: ../common/utility.cpp:87:34: error: 'log2' is not a member of 'std'

mgdudas, 2016-02-19:
Have an issue compiling firmware from the latest commit.

Compiling utility.cpp
../common/utility.cpp: In function 'float mag2_to_dbv_norm(float)':
../common/utility.cpp:87:34: error: 'log2' is not a member of 'std'
  constexpr float mag2_log2_max = std::log2(mag2_max);
                                  ^
../common/utility.cpp:87:34: note: suggested alternative:
In file included from /usr/include/newlib/c++/4.9/cmath:44:0,
                 from /usr/include/newlib/c++/4.9/random:38,
                 from /usr/include/newlib/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/newlib/c++/4.9/algorithm:62,
                 from ../common/utility.hpp:28,
                 from ../common/utility.cpp:22:
/usr/include/newlib/math.h:301:15: note:   'log2'
 extern double log2 _PARAMS((double));
               ^
../common/utility.cpp:93:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../chibios/os/ports/GCC/ARMCMx/rules.mk:165: recipe for target 'build/obj/utility.o' failed
make[1]: *** [build/obj/utility.o] Error 1
Makefile:69: recipe for target 'baseband/build/baseband.elf' failed
make: *** [baseband/build/baseband.elf] Error 2

mgdudas, 2016-02-19:
Sorry.... more info.

AMD64 Ubuntu 15.10 fully updated -- no known updates to my libraries or compiler since last successful build of a9451c1

jboone, 2016-02-19:
Sorry about that. It looks like a missing #include due to my shifting around code a bit. I have a fix ready to push (along with some other, unfinished changes), but in the meantime, you should be able to compile by adding "#include <cmath>" to firmware/common/utility.cpp, right after "#include <cstdint>".

I am a bit puzzled as to why I'm not having the issue, other than maybe some of my other changes addressed the missing header indirectly...

jboone, 2016-02-19:
Committed in d6d1f6b8af38d99e0c448b4df75dbf670db77ba5 and pushed, along with some other, incomplete changes. For instance, don't expect the screen capture button to work yet. :-)

mgdudas, 2016-02-19:
Thanks, but sadly, I have a similar issue with d6d1f6b 

Compiling utility.cpp
../common/utility.cpp: In function 'float mag2_to_dbv_norm(float)':
../common/utility.cpp:88:34: error: 'log2' is not a member of 'std'
  constexpr float mag2_log2_max = std::log2(mag2_max);
                                  ^
../common/utility.cpp:88:34: note: suggested alternative:
In file included from /usr/include/newlib/c++/4.9/cmath:44:0,
                 from /usr/include/newlib/c++/4.9/random:38,
                 from /usr/include/newlib/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/newlib/c++/4.9/algorithm:62,
                 from ../common/utility.hpp:28,
                 from ../common/utility.cpp:22:
/usr/include/newlib/math.h:301:15: note:   'log2'
 extern double log2 _PARAMS((double));
               ^
../common/utility.cpp:94:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../chibios/os/ports/GCC/ARMCMx/rules.mk:165: recipe for target 'build/obj/utility.o' failed
make[1]: *** [build/obj/utility.o] Error 1
Makefile:69: recipe for target 'baseband/build/baseband.elf' failed
make: *** [baseband/build/baseband.elf] Error 2

jboone, 2016-02-19:
Ahhhh. You're using GCC 4.9. I missed that. While I would not call GCC 4.9 unsupported, I recently moved to building with GCC 5.2 because it gives a significant improvement in code size -- enough that I can fit all the baseband code into the 32K available on the Cortex-M4F processor (as I have it configured currently).

I'm using perhaps too-aggressive standard library features, specifically that std::log2 is constexpr (computed at compile time if given constant arguments). GCC 4.9 might not allow that, but 5.2 does, even though that doesn't appear to be part of the C++11 standard. Maybe I should turn off GCC extensions... But then, my code would be less clear as to where my magic constants come from. Hmmm...

Would you be amenable to installing [GCC-ARM-Embedded 5.2](https://launchpad.net/gcc-arm-embedded/+announcement/13780), or would you prefer to make a case for continuing to support 4.9?

Alternatively, [firmware builds via Travis-CI](https://travis-ci.org/sharebrained/portapack-hackrf) are made available every time I push to the GitHub repo. URLs are of the form:

`https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/<travis_job_number>/<travis_job_number>.1/firmware/portapack-h1-firmware-<first_seven_characters_of_commit_hash>.tar.bz2`

For instance, the latest Travis build number is "73" and the commit hash is "d6d1f6b". Complete URLs are posted to Freenode #portapack after every build job -- pass or fail.

jboone, 2016-02-19:
OK, OK, putting those constants in didn't feel as dirty as expected. :-) New commit daeb542086533230f5afb37e806f086beed97127 is pushed. I hope it addresses your issue, and that you don't run out of baseband RAM with GCC 4.9. I'm pretty sure you won't...

mgdudas, 2016-02-19:
Hey Jared,

Actually, I am on 5.2:
gcc (Ubuntu 5.2.1-22ubuntu2) 5.2.1 20151010

Whatever you did resolved my issues though, daeb542 built correctly.

How can I check how much baseband RAM is being utilized? Is it the info under Debug -> Memory?

If so, here's my numbers running the new build:

M0 Core Free Bytes: 54024
M0 Heap Fragmented Free: 1032
M0 Heap Fragments: 11

jboone, 2016-02-19:
It's complicated. The way I have things divided right now, the M4 executes baseband code from a 32K block of RAM, and the stack and heap live in a much larger block of RAM. Using two distinct regions of RAM allows the M4 to execute code and fetch data simultaneously, improving performance. [More details](https://github.com/sharebrained/portapack-hackrf/wiki/Firmware%20Architecture) are in the Wiki.

The values you're seeing in Debug->Memory are for the application core (M0). The amount of baseband code RAM used is reported at compile time. The amount of baseband heap and stack used is not reported via the UI at this point, as I have not rigged up a pair of intra-core messages that the application core can use to request the data from the baseband core.

closed 2016-02-19 by jboone