sharebrained/portapack-hackrf #80: dc removal

ZeroChaos-, 2016-03-24:
gqrx has a feature for dc removal, if it's sane it might be worth killing the dc spike in the fft

irc chatter suggests this is a "lie to the user to make them stop complaining" but this bug sure looks like a complaint to me, so maybe they are right :-)

jboone, 2016-04-20:
See comment in #79. In the audio receiver modes, there is no visible DC spike, as offset tuning is used behind the scenes. This is only a visible issue in the SPEC mode. The DC spike does impact sensitivity in ERT mode, and probably has a less obvious impact even with offset tuning. So it would be good to address this at a hardware configuration level if possible, and at the software level if all other options are exhausted.

KC5CQW, 2016-05-04:
I believe the internal "birdies" from clock harmonics are being confused with DC spikes. Yes, they are very annoying but not a coding bug; rather, a hardware one.
The shield kit from nooelec.com (with the P30 header BTW) helps mitigate some of the birdies but not all. Experimentation with EMI shielding is in order. Data lines (PCB traces) may need ferrites or capacitor bypassing, I will also ask on the HackRF GitHub for any ideas.