sharebrained/portapack-hackrf #82: Receiver Menu Navigation

KC5CQW, 2016-04-29:
The UI is no longer as intuitive after the addition of the record feature. It is more difficult to navigate in the up/down range and a left/right sequence must be performed to reach the target option. NFM bandwidth is a good example. The cursor will skip over the BW and highlight record. Several L/R U/D back steps must be used to get back to the BW tab.

jboone, 2016-04-29:
This is due to the poor navigation handling algorithm. (I'm surprised you can reach the BW field, I never could.) The algorithm will be improved before the capture/record feature is released, or the UI will be revised to make the poor navigation a non-issue again.

Beware running the latest builds, there's a lot of other bugs in there still. :-)

KC5CQW, 2016-04-29:
But, I'm loving the new features! Great work. 

It's tricky to get to it, plus my down button is sticky. 

BTW, can you send me the part numbers for the selector wheel and LCD? I imagine that I will have worn them down before too long and would like to get replacements on hand incase they become end of life. 

That reminds me... My wife is always giving me a hard time about a cover to protect the face. Any files available for CNC or 3D printer that will mesh with the milled case?

73, Damon

KC5CQW, 2016-05-05:
Left click once the audio gain is highlighted. 
I find that I sometimes have to hit the reset once or twice to clear buggy behavior. 
If we don't run them, how will we find bugs? :-p

jboone, 2016-05-05:
I didn't say you shouldn't run test builds! :-)

jboone, 2016-05-05:
Ha, got it! Wow, that's a pain in the ass. I have to navigate from "NFM", along the *top* row of icons (so as to avoid changing the "option" view from "BW" to "AMP" or whatever), down to audio level, then down arrow.

In any case, that sucks, and I need a new navigation solution.

This paper was an interesting read on solving a very similar problem, DVD navigation:
https://www-users.cs.york.ac.uk/susan/bib/ss/z/zb05-dvd.pdf

Unfortunately, they were more interested in defining a means to prove an algorithm is good, more than offering a good algorithm. So back to searching...

jboone, 2016-05-05:
More on potential fixes: Web browsers implement "[Spatial Navigation](https://en.wikipedia.org/wiki/Spatial_navigation)", which is navigating between elements on a Web page using arrow keys. Following some links, I find the latest implementation of WebKit's Spatial Navigation algorithm in [a single file](http://trac.webkit.org/browser/trunk/Source/WebCore/page/SpatialNavigation.cpp). It's pretty easy to follow. I think I will implement some or all of that approach and see if it improves matters.

jboone, 2016-05-08:
Well, the code from WebKit didn't work out. In operation, it is really no better than what I have now. I'm looking at some tweaks to my existing code that might improve things.

jboone, 2016-05-08:
I've improved the algorithm in commit ab73d6155eaa4cd80dc8360172020f04a8c62aaf. Build should pop out of the Travis-CI toaster in a few minutes. Let me know if you like it -- if it seems like an improvement to you.

KC5CQW, 2016-05-09:
Mo'bettah! Works great (like it used to).

jboone, 2016-05-09:
Excellent. I'm calling it closed, then.

closed 2016-05-09 by jboone