sharebrained/portapack-hackrf #85: turn display to horizontal

lowiess, 2016-05-22:
Would it be possible to turn the display horizontal to get a better resolution ?

jboone, 2016-05-23:
This would be very nice, however the hardware prevents it from being practical. There are a few reasons:

0) The HackRF does not expose a true video interface, so I selected an LCD panel with an internal frame buffer.
1) The LCD panel has a built in scrolling function, but it only scrolls in the long dimension of the screen.
2) There is not enough memory in the HackRF to buffer and repaint the screen to accomplish scrolling in the other direction.

It may be possible to read individual lines from the LCD and write them back in a new line position, without requiring a lot of the processor's RAM. However, I am almost certain the performance will be poor, and the screen will tear and look bad during the scrolling operation.

KC5CQW, 2016-05-24:
Please see my commentary in #86 about "zoom" fit to screen.

lowiess, 2016-05-24:
OK let drop this feature request😉