sharebrained/portapack-hackrf #86: make de bandwith adjustable in SSB

lowiess, 2016-05-22:
Would it be possible to make the bandwith adjustable in AM an SSB  modes

jboone, 2016-05-23:
It's definitely possible, and at least for now, would work like the narrowband FM bandwidth adjustment.

I will admit, I don't do much HF stuff yet -- I have no decent antennas and live in a very noisy place. Are there a few specific bandwidths, or example software or radios you could point to that I could use as references for implementing this?

argilo, 2016-05-23:
For ore thing, a CW mode with a narrow filter would be useful. I've prototyped that here:

https://github.com/sharebrained/portapack-hackrf/compare/master...argilo:add-cw-mode?expand=1

I used GNU Radio's filter designer to get the narrowest 64-tap filter I could, centred at +700 Hz.

argilo, 2016-05-23:
By the way, I intentionally set the `pass_frequency_normalized` and `stop_frequency_normalized` a bit high, since otherwise the passband extends past the right edge of the waterfall.

KC5CQW, 2016-05-24:
Take a look at Flex Radio Systems... They are kind-of the ones that kicked off serious SDR for HF ham radio and government use. 
Being able to adjust the BW in SSB/CW in 10, 50, 100 & 1k steps would be a great start. Sharp cut-offs outside the "narrowed" passband and an adjustable notch filter for all modes would prove useful.
AM might not prove as useful... 6k, 9k & 12k BW "might" work for HiFi audio reception.

I would like to see a "zoom" feature to expand the viewable BW to fit the LCD screen for better viewing.
Note: The usual trade-off is lower waterfall resolution for wider screen views due to CPU effort. 

Consider adjustments for low-pass, high-pass and band-pass (pass bandwidth).
These can be located in a submenu for DSP, filters, audio etc.
Take a look at the operation manuals for one of the higher end HF transcievers (Yaesu FTDX-9000) and also test equipment like Siglent O-Scopes, Spec. Ans., for ideas on menu implementation and such.

KC5CQW, 2016-05-24:
Elecraft is also a VERY GOOD company to reference for ideas with SDR, DSP menus and such.
I personally own (and built) a K2. It uses crystal hardware filters but has a basic audio DSP option. Elecraft used this DSP as a model for an outboard audio filter kit. They should have PDFs with build, operation and theory on this and all of their other equipment.

lowiess, 2016-05-24:
Hello all

SSB and speech in general Wil fit is 300-3000 Hz thus 2700 bandwidth
And is most radio radio you Wil find filters with 2.4kc and 1.8 kc

CW a good start is 500 Hz and 250 Hz

A good source is the implementation of elecraft kx3 which is a complete SDR

Kind regards Frans

jboone, 2016-07-05:
Recent commits separated the code of the now-many baseband modes. So each baseband mode has the entire 32KBytes available for code. This opens up code space for the M4 (which has hardware floating point) to do filter coefficient calculation, instead of forcing the poor M0 core to do that work. There's also enough SPI flash that coefficients could be pre-calculated on a host computer and compiled in, but seems like a clunky and coarse approach vs. dynamically calculating coefficients based on user input.

lowiess, 2016-07-05:
Hello Jared,

If I understand well the adjustable bandwith might be possible ?
In that case there is no need for predefined bandwiths
Adjustable for 4.5 khz to 100 hz is fine for all modes

73 Frans PE0F

argilo, 2016-07-05:
Calculating coefficients is definitely a better approach. No point wasting precious memory on tables. I only added the CW table as a quick hack.