sharebrained/portapack-hackrf #88: Question Before Buying

gh0stshell, 2016-06-26:
Can you record a signal and then play it back using the portapack H1 and HackRF?

jboone, 2016-06-26:
You can record a signal (500kHz bandwidth) to a good-quality microSD card, but only in the pre-release firmware. I keep saying I'm going to release the firmware next week, but next week hasn't come yet. :-( Anyway, if you install the pre-release firmware, it should work fine. I know of several people who are using it with no complaints (other than some user interface quirks).

As for replay, at the moment you will need to use a computer for that. The next firmware release will incorporate replay of files from SD card. I very much want to finish that release before Black Hat USA and DEF CON this year -- so five weeks or so.

gh0stshell, 2016-06-27:
OK, what are the steps to upgrade firmware? I did not find a guide/steps with just searching. If that feature will be available in the future I will order my hackrf and portapack H1.

dc1rdb, 2016-06-27:
@gh0stshell , you might want to check the wiki https://github.com/sharebrained/portapack-hackrf/wiki , it contains detailed steps to install continuous integration builds of the firmware. It's very easy.

jboone, 2016-06-27:
@gh0stshell One important detail... The download URLs in the continuous build log are broken due to some Travis-CI configuration problem I was unable to resolve. A friend set up a redirector to the correct URL for the latest build. See issue #73 for more details.

gh0stshell, 2016-06-28:
The best place I could find both was Hackerwarehouse. They are out of HackRF's but have portapacks. I bought both and while I wait for the HackRF to be back in stock to ship me one I asked them to send the portapack so I can check it out and get the firmware we talked about on it. I would be happy to be a beta tester if you need/want on.

argilo, 2016-06-28:
@gh0stshell The Portapack firmware goes onto the HackRF, so you'll have to wait until you have that in hand.

KC5CQW, 2016-06-29:
_"I very much want to finish that release before Black Hat USA and DEF CON this year -- so five weeks or so."_
Any chance this 'release' would also allow TX?

gh0stshell, 2016-06-30:
Super excited, both have arrived!!!!!!!!!!!!!!!!!!

gh0stshell, 2016-06-30:
@KC5CQW, yes that was my original question.

jboone, 2016-06-30:
@KC5CQW The replay feature would transmit, yes. If you're meaning something more specific (e.g. transmit of analog audio modes), we will see...

gh0stshell, 2016-07-08:
Which firmware should I install? I am done playing with the HackRF connected to a computer.

jboone, 2016-07-08:
If you want to be brave (and deal with minor bugs and unfinished features), please install the firmware at the URL discussed in issue #73. Otherwise, I recommend the latest released firmware, available in the ["releases" portion of this repository](https://github.com/sharebrained/portapack-hackrf/releases).

gh0stshell, 2016-07-10:
i will live, installing that firmware shortly

gh0stshell, 2016-07-10:
link does not work

jboone, 2016-07-10:
Looks like the "munchkins" redirector no longer works, because CMake now places the build artifacts in a "build" subdirectory. The correct URL can still be obtained from the Travis-CI log by applying information in the first comment of issue #73. The correct URL for the latest build is: https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/111/111.1/build/firmware/portapack-h1-firmware-db700b1.tar.bz2

gh0stshell, 2016-07-10:
great, doing that now

gh0stshell, 2016-07-10:
i want to enter a Ghz freq, how do I do that on the portapack? I key fob I own runs in the 2Ghz range and before I tinker with the signal I want to see what makes the signal activate

jboone, 2016-07-10:
This is not the best avenue for ongoing individual support. Please e-mail me at info@sharebrained.com, or join us in IRC: Freenode #portapack.

closed 2016-07-10 by jboone

Redouan29, 2016-07-21:
All the Portapack's are available from the US. Do anyone know a good vendor from inside the EU? (avoiding taxes and other duties).

jboone, 2016-07-24:
@Redouan29 Not yet. If you have a vendor you prefer, please e-mail me at the address mentioned two comments earlier.