sharebrained/portapack-hackrf #97: Restoring HackRF Firmware

taintdluv2k4, 2016-07-28:
I followed the directions in the section titled, "Restoring HackRF Firmware" and received the following output:
No DFU capable USB device found

While in this state, or in Portapak mode, and even running in "HackRF Mode", no new devices show up for the "lsusb" command either...

jboone, 2016-07-28:
The HackRF will only appear as a DFU device if you hold the DFU button while plugging in USB or pressing and releasing the RESET button.

The PortaPack firmware does not use USB and will not show up at all on a host computer.

You should see something with lsusb after switching from PortaPack to HackRF mode. Let's get the easy things out of the way -- try different cables, different USB ports on your computer, reboot your OS. A surprising number of these issues are caused by a bad cable or connector, or the operating system being in a bad mood. Also, check that your OS power management isn't interfering. For instance, this HackRF issue: https://github.com/mossmann/hackrf/issues/243

Let me know what you discover.

taintdluv2k4, 2016-07-28:
Oh, the shame... it was just a bad cable.

jboone, 2016-07-28:
No worries. You're definitely not alone. ;-)

closed 2016-07-28 by jboone