sharebrained/portapack-hackrf #102: New Stable Release Build Issues

gh0stshell, 2016-08-18:
After flashing the latest release under releases I have some odd issues when entering the freq number and also a few times I cannot get to the top menu buttons like back and when these issues happen I have to reset the device to get working again. If I go back to the build db700b1 I do not have the same issues I do with the latest release build.

jboone, 2016-08-21:
@gh0stshell Is this via the touch interface, or using the mechanical controls?

jboone, 2016-08-31:
@gh0stshell Any extra insights you could share?

gh0stshell, 2016-08-31:
Touch screen, if I go to enter a freq say 315, when pressing 3 it enters 3, when pressing 1 it enters 3-4 1 numbers, when I re-flashed db700b1 when I enter 315 it enters 315 with no extra numbers. As for the menus, it has a few times where when I use the scroll wheel to go to the top options I can never get to the back or any of the menus at the top above the freq section.

gh0stshell, 2016-08-31:
After resetting I can get to the menus normally, but the multiple numbers when entering a freq still happens so I went back to db700b1

jboone, 2016-08-31:
@gh0stshell The multiple numbers is likely due to lighter debouncing of detected screen touches. I have noticed this a few times, to a lesser degree, on a couple of my units. It's a regression I will address for the next release.

As for the top options being inaccessible, I'm not sure how to recreate this. Firstly, is this in the audio receiver mode? Do you see the menu issue with the db700b1 firmware? Can you identify a minimal set of steps required, from a fresh reset, to recreate the issue? Or is it something that seems to happen at random? Are there any general trends or patterns you can identify? Or if you can recreate it, even sporadically, perhaps you can record a video example you can share?

gh0stshell, 2016-08-31:
Yes, all the issues I have had happen with the latest stable was in the receiver screen. I do not get them with the db700b1 build so I have gone back to that build as it works very well for me even though it technically a beta build. The top menu issue is random for me and I do not notice it is happening until I try to go back a menu.

gh0stshell, 2016-08-31:
I will try to shoot a vid when it happens, I will flash the latest stable tonight when I have my gear.

gh0stshell, 2016-09-01:
New issue I have come across which may fix the enter of freq while in receive mode, cannot get the touch setup to work, will not recognize my input.

aaronkondziela, 2016-09-02:
Just to throw in, I've also had the same trouble where the mechanical controls seem to get into a funky state, and I can no longer navigate without a reset. Can't comment on the touchscreen aspect, as I don't use it often. It does seem to me like some kind of debouncing issue, on my unit specifically with the center button. (Maybe I should toss the scope on it, see how much bounce there is?) Almost feels like the software gets into a funny state where it thinks the button remains held down or something. Also note that it seems random, and did not start after a particular firmware; has been present through a few versions.

jboone, 2019-03-21:
Some code was added to deal with mechanical wheels being non-ideal devices, including ignoring secondary key presses. Imminent new release may help. Closing...

closed 2019-03-21 by jboone