sharebrained/portapack-hackrf #103: Latest Build Address

gh0stshell, 2016-08-18:
Do any of the newest builds have the TX option and if so what is the latest link to download it so I can try the build? 

I see listed in travis-CI under current as of 2 days ago there is a new build, but do not know if it has TX as I am having issues location the download URL as listed in the steps for issue #73

portapack-h1-firmware-8333789.tar.bz2

gh0stshell, 2016-08-19:
I see the re-director is working and pulled the latest build from that

http://munchkins.net/portapack

jboone, 2016-08-21:
@gh0stshell I have some prototyped TX features in my local branch of the repository. None of them are usable except for my testing. But, they are working! So I can promise there will be interesting TX features in the next release. And the next release will be much sooner than the five months it took for the last release...

closed 2016-08-21 by jboone