sharebrained/portapack-hackrf #104: firmware build instructions don't work?

mgdudas, 2016-08-31:
Hi,

I noticed about 4-5 months ago, my build environment completely died, and I was no longer able to build firmware. At the time, and ever since I figured it was either a temporary break, or my environment had completely broken (which had happened before as well)

But, today I finally build out a completely fresh Ubuntu 16.04 VM, and I still can't get it to build anything.

When I grab a fresh clone of the git repo, after having installed all other dependencies, I try to run 'make' in the firmware directory and get this error:

make: *** No targets specified and no makefile found.  Stop.

Since the firmware build instructions haven't seemed to change in quite some time, I really have no idea what the problem is. Has something changed, or am I just failing at basic compiling?

Thanks!

jboone, 2016-08-31:
@mgdudas No, you're not failing. I'm failing at updating the documentation. The firmware build now uses CMake (2.8.10 or newer). The instructions, if written somewhere appropriate, would look like this:

cd <portapack_source_directory>
mkdir build
cmake ..
make

...and it should pop out, after much pomp and circumstance, a binary you can load in the usual manner (via hackrf_spiflash). Or you may put the HackRF into DFU mode and do "make program".

I hope that helps.

(I'm keeping this issue open as a reminder to me to update the blasted docs.)

jboone, 2016-08-31:
@mgdudas Correction, I forgot that a simple "make" doesn't produce the "firmware/portapack-h1-firmware.bin" file. That comes during "make program". So that needs improving, too...

mgdudas, 2016-08-31:
@jboone Great, I'm not insane! You know, I had noticed the cmake files and such before, and at one point thought I had gotten somewhere, but I guess I was in the wrong directory.

Ultimately though, running make program, I still get an error (because of dfu-util 0.8 I think):

Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
dfu-util: WARNING: Runtime device already in DFU state ?!?
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        20484 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
dfu-util: error resetting after download
firmware/CMakeFiles/program.dir/build.make:57: recipe for target 'firmware/CMakeFiles/program' failed
make[3]: *** [firmware/CMakeFiles/program] Error 74
CMakeFiles/Makefile2:130: recipe for target 'firmware/CMakeFiles/program.dir/all' failed
make[2]: *** [firmware/CMakeFiles/program.dir/all] Error 2
CMakeFiles/Makefile2:137: recipe for target 'firmware/CMakeFiles/program.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/program.dir/rule] Error 2
Makefile:131: recipe for target 'program' failed
make: *** [program] Error 2

Then, realizing I still have dfu-util 0.8 installed, I try the 'workaround' listed on the build page, but that is also incorrect, because the file locations have changed.

From build dir:

hackrf_one_usb_ram.dfu is now in ../firmware
portapack-h1-firmware.bin is now in ./firmware

Even after fixing the paths though, I get the following:

dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
dfu-util: WARNING: Runtime device already in DFU state ?!?
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        20484 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
dfu-util: error resetting after download
File size 422720 bytes.
hackrf_open() failed: HACKRF_ERROR_NOT_FOUND (-5)

:-(

jboone, 2016-08-31:
@mgdudas You shouldn't need to fix any paths. The fact that you had to is an indication something else is wrong.

dfu-util 0.8 is fine, I've been using it for months with no (real) problems, other than it complains a lot about things that aren't, in practice, errors.

The fact that the DFU "Download [===...]" completes is a sign that your hackrf_one_usb_ram.dfu was in the right place to begin with.

You're very close. In the last try, think your build is fine. It looks like the HackRF mode USB interface isn't enumerating. One of several possible causes: bad USB cable, a virtual machine guest that is stealing the USB device or otherwise interfering with USB enumeration, or the delay between DFU download and expecting the HackRF to enumerate (the "sleep 1") is too short.

jboone, 2016-08-31:
@mgdudas Also, if you "VERBOSE=1 make" or "VERBOSE=1 make program", you'll see lots of extra detail that may show exactly what's going wrong.

mgdudas, 2016-08-31:
Well, the first time I run 'make program' it knows where the files are, and does complete the download. But the 2nd time, I'm using the dfu-util 0.8 workaround listed at the bottom of the build page, and that exact command no longer has the correct paths.

As for the hackrf itself, hackrf_info shows it just fine, and dfu-util lists it correctly when in dfu mode. I'll try a different cable and see what I can do.

While I'm at it... I've attached a file with the log of 'VERBOSE=1 make program' if that helps determine what the issue is.

Thanks!
[make-program-failure.txt](https://github.com/sharebrained/portapack-hackrf/files/447911/make-program-failure.txt)

mgdudas, 2016-08-31:
Ultimately, after trying a new cable, make program still fails, but running the workaround command (with corrected paths from the build dir) it has finally properly programmed it.

dfu-util --device 1fc9:000c --download ../firmware/hackrf_one_usb_ram.dfu --reset
hackrf_spiflash -w ./firmware/portapack-h1-firmware.bin

So, seems it's a combination of things. :-)

Thanks again for your help.

jboone, 2016-08-31:
@mgdudas So weird. We're running exactly the same code. Your log diverges from mine when yours says "dfu-util: error resetting after download". I'm guessing from your paths you're on Linux, possibly Ubuntu? Oh, yeah, you said that initially. It seems like your CMake or dfu-util is treating the reset failure as a build failure, and mine isn't. What version of CMake are you running? I'll also check to see that there isn't a rev of dfu-util or CMake, beyond what I'm running on Mac OS.

jboone, 2016-08-31:
@mgdudas If you remove the "--reset" from the DFU step in the CMake file, does everything start working?

mgdudas, 2016-08-31:
I assume you mean remove it from firmware/CMakeLists.txt ?

I'm running:
Ubuntu 16.04.1 LTS
cmake 3.5.1
dfu-util 0.8

After removing --reset from firmware/CMakeLists.txt line 62 and compiling it all again, 'make program' produces the following (truncated) output, and it does appear to work:

<truncated>
dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
dfu-util: WARNING: Runtime device already in DFU state ?!?
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        20484 bytes
Download done.
dfu-util: unable to read DFU status after completion
File size 422720 bytes.
Erasing SPI flash.
Writing 422720 bytes at 0x000000.
[100%] Built target program

One other random note.... on the about screen, it says HackRF CPLD BAD... I don't recall it saying BAD previously. Not sure if it's related.

jboone, 2016-08-31:
@mgdudas It's not related, and I'd expect it to say "BAD" right now. It indicates that the bitstream flashed into the HackRF CPLD does not match the one that the PortaPack is using. I need to separate things out, so I can validate the flashed bitstream is correct for the HackRF firmware, while potentially using a different bitstream (now the case in the repo HEAD) for PortaPack mode.

I'm going to remove the "--reset" from the firmware/ CMake file.

jboone, 2016-08-31:
By the way, I've updated the Wiki:

 https://github.com/sharebrained/portapack-hackrf/wiki/Building-Firmware

Let me know if it matches your experience.

jboone, 2016-08-31:
Pushed the DFU --reset change. Let me know if, when building from a fresh build/ directory, it still programs the device OK.

mgdudas, 2016-08-31:
OK, I can confirm after wiping my entire existing directory and starting over from scratch. Following the instructions on the current wiki page works flawlessly.

Thanks for all your help.

jboone, 2016-08-31:
Excellent. Thank *you* for your help, and letting me know about this issue. Closing the issue.

closed 2016-08-31 by jboone