sharebrained/portapack-hackrf #106: HackRF mode missing from screen, cannot update firmware

FT847, 2016-12-02:
I must have installed an older version of the Portapack firmware. Now, I cannot update the firmware since the HackRF menu item no longer exists. 
I cannot even reset using dfu_hack_one application. 
My OS is Windows 10 home. 

[image]
[image]
[image]

Thank you in advance for your assistance.

jboone, 2016-12-03:
@FT847 I've contacted you via e-mail, but let me also write the answer publicly so it can help other Windows users -- and serve as a reminder to me to update the Wiki with this process.

Uninstall existing copies of the PortaPack firmware tool, to ensure a clean slate.

Find the latest PortaPack firmware under the "Releases" section of this repository.

Download and install "PortaPack.Firmware.Tool" executable.

Inside the PortaPack program group, you should find a program (actually a batch file) called "HackRF Mode via DFU". Run that program and follow the instructions.

Once you press a key, the HackRF firmware will be loaded into the memory of the HackRF. It will not persist, so do not unplug or reset the HackRF.

Now, try to use either of the "Flash PortaPack Firmware" or "Flash HackRF One Firmware" tools. Once you've gone through that process, you should be good to go with whichever firmware you chose to install.

FT847, 2016-12-05:
Jason. I loaded 20160728 and got the result as shown in the "DFU-Flash.jpg". Any ideas? -----Original Message-----
From: Jared Boone 
Sent: Dec 2, 2016 7:37 PM
To: sharebrained/portapack-hackrf 
Cc: FT847 , Mention 
Subject: Re: [sharebrained/portapack-hackrf] HackRF mode missing from screen, cannot update firmware (#106)

@FT847 I've contacted you via e-mail, but let me also write the answer publicly so it can help other Windows users -- and serve as a reminder to me to update the Wiki with this process.
Uninstall existing copies of the PortaPack firmware tool, to ensure a clean slate.
Find the latest PortaPack firmware under the "Releases" section of this repository.
Download and install "PortaPack.Firmware.Tool" executable.
Inside the PortaPack program group, you should find a program (actually a batch file) called "HackRF Mode via DFU". Run that program and follow the instructions.
Once you press a key, the HackRF firmware will be loaded into the memory of the HackRF. It will not persist, so do not unplug or reset the HackRF.
Now, try to use either of the "Flash PortaPack Firmware" or "Flash HackRF One Firmware" tools. Once you've gone through that process, you should be good to go with whichever firmware you chose to install.

FT847, 2016-12-11:
Issue resolved, thanks again for your quick response.

closed 2016-12-11 by FT847