sharebrained/portapack-hackrf #107: Build failure - spectrum_collector.cpp:22:

aernan, 2017-01-20:
Gingerbread-Hollicaust:build aernan$ make firmware
[  0%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/spectrum_collector.cpp.obj
In file included from /Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.hpp:28:0,
                 from /Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:22:
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/block_decimator.hpp: In instantiation of 'constexpr BlockDecimator<T, N>::BlockDecimator(size_t) [with T = std::complex<short int>; unsigned int N = 256u; size_t = unsigned int]':
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.hpp:48:66:   required from here
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/block_decimator.hpp:37:23: warning: missing initializer for member 'std::array<std::complex<short int>, 256u>::_M_elems' [-Wmissing-field-initializers]
  ) : factor_ { factor }
                       ^
In file included from /Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:22:0:
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.hpp:54:58: warning: missing initializer for member 'std::array<std::complex<float>, 256u>::_M_elems' [-Wmissing-field-initializers]
  std::array<std::complex<float>, 256> channel_spectrum { };
                                                          ^
In file included from /Users/aernan/Hardware/portapack-hackrf/firmware/common/portapack_shared_memory.hpp:28:0,
                 from /Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:28:
/Users/aernan/Hardware/portapack-hackrf/firmware/common/message_queue.hpp:83:22: warning: missing initializer for member 'Mutex::m_queue' [-Wmissing-field-initializers]
  Mutex mutex_write { };
                      ^
/Users/aernan/Hardware/portapack-hackrf/firmware/common/message_queue.hpp:83:22: warning: missing initializer for member 'Mutex::m_owner' [-Wmissing-field-initializers]
/Users/aernan/Hardware/portapack-hackrf/firmware/common/message_queue.hpp:83:22: warning: missing initializer for member 'Mutex::m_next' [-Wmissing-field-initializers]
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp: In instantiation of 'typename T::value_type spectrum_window_hamming_3(const T&, size_t) [with T = std::array<std::complex<float>, 256u>; typename T::value_type = std::complex<float>; size_t = unsigned int]':
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:142:79:   required from here
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:114:2: error: non-constant condition for static assertion
  static_assert(power_of_two(s.size()), "Array size must be power of 2");
  ^
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:114:2: error: 's' is not a constant expression
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:115:37: error: 's' is not a constant expression
  constexpr size_t mask = s.size() - 1;
                                     ^
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp: In function 'typename T::value_type spectrum_window_hamming_3(const T&, size_t) [with T = std::array<std::complex<float>, 256u>; typename T::value_type = std::complex<float>; size_t = unsigned int]':
/Users/aernan/Hardware/portapack-hackrf/firmware/baseband/spectrum_collector.cpp:118:1: warning: control reaches end of non-void function [-Wreturn-type]
 };
 ^
make[3]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/spectrum_collector.cpp.obj] Error 1
make[2]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2

jboone, 2017-02-22:
@aernan What version of G++ are you using?

jboone, 2017-08-12:
Closing. This is most likely due to my rather aggressive updating of G++ to the latest version. I will not repent! :-)

closed 2017-08-12 by jboone