sharebrained/portapack-hackrf #108: HackRF One FW update

KC5CQW, 2017-02-18:
Include the latest FW release 2017.02.1 to replace the 2015 version.

jboone, 2017-02-22:
Firmware is incorporated on this testing branch: https://github.com/sharebrained/portapack-hackrf/tree/working_20160728.1

Please let me know how it works for you. Cursory tests worked fine for me.

jboone, 2017-02-22:
Also addressed in commit 66d66b57ab8ef89e00a3491532af8c620f20a9e8 on master.

jboone, 2017-03-24:
Re-opening for futher testing and feedback. Builds for testing are available here:

https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/140/140.1/build/firmware/portapack-h1-firmware-66d66b5.zip

https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/140/140.1/build/firmware/portapack-h1-firmware-66d66b5.tar.bz2

jboone, 2017-08-12:
2017.02.1 is distributed in the newest PortaPack firmware releases and is working, apparently without issue.

closed 2017-08-12 by jboone