sharebrained/portapack-hackrf #112: lcd model / part number

gigawatts, 2017-04-22:
Can you please provide a part number and supplier for the LCD, I need to find a replacement (LCD cracked, digitizer no longer works). Actually, is it possible to just replace the digitizer? I am pretty sure that is the only part that is broken.

I didn't see a BOM anywhere, sorry if I overlooked it.

sck-nogas, 2017-04-24:
Last year, I cracked my LCD screen, and Jared in August 2016 said... "The LCDs
come directly from China, and have a unique pin-out. They do not use the SPI
interface. I do sell replacement LCDs at my cost ($5 + shipping),"

It was real easy to replace.

Scott

gigawatts, 2017-04-27:
Thanks for the info, I'll shoot Jared an email.

gigawatts, 2017-06-04:
Jared was also able to sell me a replacement LCD, which was very easily installed. Thank you again for the tip @sck-nogas

jboone, 2017-08-12:
While I sell LCDs to those who contact me, you can find the LCDs by these rules:

PortaPack revision before 20170522 --> Kingtech DW0240A2BZ (16-bit 8080-I MCU interface, IM[3:0] = 0b0001) with resistive touch screen.
PortaPack revision 20170522 (and newer, if that should happen?) -> EastRising ER-TFT024-3 with resistive touch screen.

This information has been added to the "Hardware" section of the wiki.

closed 2017-08-12 by jboone