sharebrained/portapack-hackrf #113: TCXO

ncoop23, 2017-08-09:
How do I go about installing a TCXO on the HackRF with a PortaPack installed?  It appears that the header is used on the board.  Is there a way to kludge this

Possible to integrate a TCXO on the next hardware version?

ncoop23, 2017-08-09:
Worst case is something powered over USB that puts the clock signal into the clkin SMA connector but something in the box would probably be better.

jboone, 2017-08-12:
@ncoop23 I would dead-bug the TCXO onto the PortaPack. The pins are sufficiently exposed that it shouldn't be too difficult. Be mindful, however, of total power consumption, as USB limits draw on the VBUS to <500 mA, and has strict limits around how much current you can draw when first connecting to USB power. Also, you'll want to match the signal requirements for the Si5351C-B input, and ensure you don't plug anything into the reference clock input SMA (which risks damaging whatever TCXO you've attached).

closed 2017-08-12 by jboone