sharebrained/portapack-hackrf #114: LCD read broken (at least on old version)

furrtek, 2017-08-11:
It seems that the LCD read signal polarity change in 751ae9 causes data returned to be all bits stuck to 1.

jboone, 2017-08-11:
This appears to be due to gross negligence on my part. Fix forthcoming...

jboone, 2017-08-11:
Fix committed above. Please test and verify that it works for you. I was able to reproduce the issue and test the fix here, but apparently, I shouldn't trust myself...

furrtek, 2017-08-11:
Tested and verified, read_pixels() is now working. Thanks :)

jboone, 2017-08-12:
Excellent. Thanks for checking, and for finding this to begin with. I'm a bit surprised it worked at all. Apparently the CPLD tools picked the same LCD_WRX pin as I'd intended, by my dumb luck.

closed 2017-08-12 by jboone