sharebrained/portapack-hackrf #116: DFU fault happened in linux Ubuntu14

wog5zpf, 2017-12-15:
dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

I executed this command "dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset" and met the problem of DFU as above. Could anyone help me please.

jboone, 2018-12-16:
@wog5zpf I apologize for the very long delay.

The first two messages are OK. dfu-util complains a lot about things that are not actually errors.

The last error is a concern. First, be sure that you are holding down the DFU button on the HackRF (it is the button closer to the antenna connector) while you plug in the USB cable.

If you are using Linux, do you see the device detected in "dmesg" output? For instance:

```
[212531.109068] usb 1-2: new high-speed USB device number 83 using xhci_hcd
[212531.258097] usb 1-2: New USB device found, idVendor=1fc9, idProduct=000c
[212531.258104] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[212531.258108] usb 1-2: Product: LPC
[212531.258112] usb 1-2: Manufacturer: NXP
[212531.258116] usb 1-2: SerialNumber: ABCD
```

jboone, 2019-01-24:
Nothing heard. Closing.

closed 2019-01-24 by jboone