sharebrained/portapack-hackrf #117: Touchscreen issues

szedd, 2018-01-15:
When I try to calibrate the touchscreen, it often seems to be unresponsive. 
When it is working, it only seems to recognize the first two calibration points (upper left and right),
the calilbration fails at the third point or it starts to take screenshots and enter sleep mode before it fails.

The firmware date is August 9th 2017 (Git Commit Hash a335139)

I don't know if it really is a firmware issue or the hardware failed.

jboone, 2018-01-15:
The touchscreen calibration mode has turned out to be a bit of a mistake. It's very fussy, and only works well when using something pointed, like a stylus. I really need to either disable it, or loosen up the tolerances.

Bear in mind that this is  a resistive touch screen, which is terrible by comparison to modern smartphone capacitive touch devices... I've mostly avoided implementing touch features in the user interface because of how disappointing it is vs. modern expectations.

When you do make it through the calibration process, does touch entry (say in the receiver frequency screen) work as expected, or is that erratic?

szedd, 2018-01-15:
Touch entry doesn’t work right at all, all it does is taking screenshots and enter sleep mode, I guess it thinks I am touching the upper right of the screen all the time.
I tried using a stylus, that didn’t help either.

szedd, 2018-01-15:
I took a Video of the problem https://youtu.be/Bje9wAHSIsc
As you can see, it is not responsive when I try to enter a frequency.
In calibration mode it often takes screenshots when I push center or left and it enters sleep mode when i push on the right side.

From: Jared Boone 
Sent: Monday, January 15, 2018 5:01 AM
To: sharebrained/portapack-hackrf 
Cc: szedd ; Author 
Subject: Re: [sharebrained/portapack-hackrf] Touchscreen issues (#117)

The touchscreen calibration mode has turned out to be a bit of a mistake. It's very fussy, and only works well when using something pointed, like a stylus. I really need to either disable it, or loosen up the tolerances.

Bear in mind that this is a resistive touch screen, which is terrible by comparison to modern smartphone capacitive touch devices... I've mostly avoided implementing touch features in the user interface because of how disappointing it is vs. modern expectations.

When you do make it through the calibration process, does touch entry (say in the receiver frequency screen) work as expected, or is that erratic?

jboone, 2018-01-15:
@szedd Hmmm, yeah. I was hoping this would be something we could fix remotely. Please contact me by the e-mail address at the bottom of [my contact page](https://www.sharebrained.com/about/) and we can discuss how to get that hardware fixed.

closed 2018-01-16 by szedd