sharebrained/portapack-hackrf #120: Problem with HackRF after adding portapack

alphaomicron, 2018-03-07:
Hello Jared,
I finally got my portapack and started assembly. Once I attempted a first boot hackrf, went in dfu mode and can't reset it since then. I had updated to firmware 2017.1 and everything worked fine since I last tested. 
I followed https://github.com/sharebrained/portapack-hackrf/issues/108  but no success 
Running on either VM or VirtualBox on a windows machine

Thank you!

alphaomicron, 2018-03-07:
Any chance that might be caused by a microsd into the portapack which was straight out of the box? So somehow the portapack tried to read a firmware from the sd?

alphaomicron, 2018-03-07:
Ok this is so lame of me...and although I was aware of similar problems I thought it would never happen to me again. It was a cable problem and although I used the cable that came with the Hackrf it seems that it caused the problem. Issue closed

alphaomicron, 2018-03-07:
Nope same problem again as soon as I attach portapack :-(

alphaomicron, 2018-03-07:
Hackrf is recognized by the computer but nothing comes up on the portapack screen

alphaomicron, 2018-03-07:
Hackrf is recognized by the computer but nothing comes up on the portapack screen. I makes a quick flash and then there is nothing.
Also when attached to two different power banks there is no usb light (maybe an issue for the Hackrf repository)

jboone, 2018-03-07:
I received a lot of messages from this issue, and don't see some of them. Can you tell me what's working and what isn't? And if you'd prefer a more interactive medium, would you like to meet in Freenode #portapack?

alphaomicron, 2018-03-07:
Of course! would love to contribute in anyway I can.Yeap we can meet in #portapack

closed 2018-03-07 by alphaomicron