sharebrained/portapack-hackrf #125: TXCO Follow-up

cyrus104, 2018-06-11:
I saw your post about the TXCO having sufficient pins exposed. My question is besides needing to raise all the pins and the case not fitting 100% could I use a TXCO in J22 with double sided pins and double height pins everywhere else on the board. Will there be an issue with the TXCO and portapack using the same pins?

Thanks

TheCranston, 2018-06-11:
@cyrus104 I deadbug'd my TCXO to the bottom of the HackRF board.  Laid down some insulating tape on the adjacent side of the Portapack case to ensure if contact is somehow made that it's not electrical...  However, I believe the code to select the external clock is not active in the Portapack firmware so you'll not get the benefit of the clock unless you are in HackRF mode.

cyrus104, 2018-06-13:
ok, nice which TXCO did you use? Is there any possibility to add the external clock capability into the Portapack firmware?

TheCranston, 2018-06-13:
I purchased a TTETKLJANF-10.000000 from digikey.  It's a 0.28ppm TCXO.  It was less than $15.  The app notes suggest a 0.1uf decoupling capacitor between Vcc and GND, but I omitted that.  I'm getting a reliable 0.3ppm calibrating from a local LTE tower.  

furrtek is working on including the external clock into the havoc firmware.

cyrus104, 2018-06-16:
@TheCranston, thanks for the input I order the 0.10ppm version of the same TCXO. I should get it over the weekend or early next week. Can you post a picture of how you installed it? I'm in the process of looking at how to solder it onto the hackrf while still using the portapak. I'll just need to use it in Hackrf mode till the Havoc firmware supports it.

sgoudelis, 2018-11-14:
Hey @TheCranston can you upload a photo of the hackrf with the TCXO ?

jboone, 2018-12-16:
@cyrus104 Support for detecting and using an external 10 MHz clock reference is present in this test firmware, if you want to give it a try: https://portapack-h1-builds.s3.amazonaws.com/sharebrained/portapack-hackrf/176/176.1/build/firmware/portapack-h1-firmware-90f64a6.tar.bz2

TheCranston, 2018-12-17:
@sgoudelis, perhaps after the new year.  I'm totally swamped right now with stuff and would have to take apart the whole unit to get at it.  For what it worth, I just hot glued the TCXO to the bottom of the HackRF right next to the header pins.  Then three jumpers, power, GND, and clock output, over to the bottom of the board on those respective pins.  keep the wires short.

jboone, 2019-01-15:
Support for external clock reference, and the clock reference on new PortaPacks, has been tested and is present in the latest builds of this firmware. HAVOC has a pending pull request (update) that should similarly address this issue. I'm going to close this. If y'all have the time to test, please do. Otherwise, look for a firmware release in the next couple of days.

closed 2019-01-15 by jboone